adafruit/Adafruit_CircuitPython_BusDevice #19: explicit 'write-then-read' function for repeated start

ladyada, 2018-08-12:
exciting news - while microcontrollers let you not-send a STOP, linux doesn't allow it explicitly. instead, we must use a single call that does reads and writes

e.g. here's the pure smbus example:
```
import datetime
import Adafruit_PureIO.smbus as smbus
import time

bus=smbus.SMBus(1)

# how busdevice works now, we write the register, then read separetly. great on micros, bad on linux.
bus.write_byte(0x21, 0x0C)
print(hex(bus.read_byte(0x21)))

# works correctly on linux, single ioctl reads 1 byte from register 0x0C on addr 0x21
print([hex(i) for i in bus.read_i2c_block_data(0x21, 0x0C, 1)])
```
but, right now, bus device only has read and write. i'd basically like to add a function called write_readinto, we'll also need a matching function in the busio, but for microcontrollers it'd just call the two functions immediately one after the other. in blinka linux i'd call the special block_data command

deshipu, 2018-08-12:
I don't think this is a difference between Linux and microcontrollers, but rather between i2c and smbus. Smbus is a subset of i2c, with additional restrictions. Sometimes it works with i2c devices, but there are devices out there for which it won't work.

deshipu, 2018-08-12:
As a side note, I once tried to implement an smbus module for the micro:bit, so that you could use drivers written for rpi. Turns out that you can't implement smbus block read and block write commands correctly with the API available there. I think CircuitPython's API has a similar problem: https://github.com/bbcmicrobit/micropython/pull/376

ladyada, 2018-08-12:
totally, but linux doesnt have any pure i2c ioctls. which makes sense - like spi, theres context switching at any point, so they want to force the user to be 'transaction' based rather than holding a lock.

but, we should totally be able to do it on circuitpython since we implemented stop selection - a write with no stop, then a read with stop is the same thing?

looks like micropython is going about this a slightly different way
https://github.com/micropython/micropython/pull/4020

ladyada, 2018-08-18:
i did this! its called write_then_readinto, works great :)

closed 2018-08-18 by ladyada