adafruit/Adafruit_CircuitPython_BusDevice #39: docs link is 404'd

jepler, 2020-01-31:
The link from README.rst, https://circuitpython.readthedocs.io/projects/bus_device/en/latest/ , does not work.  This is also causing doc build failures in repositories that refer to this one via intersphinx, such as at https://github.com/adafruit/Adafruit_CircuitPython_VL53L0X/runs/419328376 .

The docs appear to exist at https://circuitpython.readthedocs.io/projects/busdevice/en/latest/ instead (note the missing underscore).

If the link is wrong and needs to be updated, it looks like the following projects in Adafruit_Circuitpython_Bundle will need to have their intersphinx links updated:

https://github.com/adafruit/Adafruit_CircuitPython_FeatherWing
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice
https://github.com/adafruit/Adafruit_CircuitPython_VL6180X
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx
https://github.com/adafruit/Adafruit_CircuitPython_TCS34725
https://github.com/adafruit/Adafruit_CircuitPython_AS726x
https://github.com/adafruit/Adafruit_CircuitPython_VCNL4010
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561
https://github.com/adafruit/Adafruit_CircuitPython_SI4713
https://github.com/adafruit/Adafruit_CircuitPython_RFM9x
https://github.com/adafruit/Adafruit_CircuitPython_VL53L0X
https://github.com/adafruit/Adafruit_CircuitPython_TSL2591
https://github.com/adafruit/Adafruit_CircuitPython_WS2801
https://github.com/adafruit/Adafruit_CircuitPython_MPL3115A2
https://github.com/adafruit/Adafruit_CircuitPython_VEML6070
https://github.com/adafruit/Adafruit_CircuitPython_MMA8451
https://github.com/adafruit/Adafruit_CircuitPython_SI5351

ladyada, 2020-01-31:
@dherrada hi dylan, would you be up to fixing the URLs in these repos? wouldn't be automated, each one needs a _ removed

dherrada, 2020-01-31:
@ladyada sure

dherrada, 2020-02-03:
All PRs have been made. I accidentally pressed the wrong button and commited to master on VL6180X and MCP230xx.

ladyada, 2020-02-03:
thats ok :)

dherrada, 2020-02-05:
@kattni (almost) all releases are in. I forgot to tag circuitpython librarians in a few of the PRs. Here are their links:

* https://github.com/adafruit/Adafruit_CircuitPython_FeatherWing/pull/53
* https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/pull/40

kattni, 2020-02-05:
@dherrada Merged. Please release those two and then we can close this issue. Thanks!

dherrada, 2020-02-06:
I've done the last 2 releases. Closing

closed 2020-02-06 by dherrada