adafruit/Adafruit_CircuitPython_BusDevice #53: adafruit_bus_device writeto(..., stop=stop) deprecated

anecdata, 2020-07-24:
> Breaking changes since 5.x:
> The stop kwarg has been removed from I2C.writeto(). If no stop is desired, then use writeto_then_readfrom.
> 

```  File "adafruit_mpu6050.py", line 168, in __init__
  File "adafruit_mpu6050.py", line 182, in reset
  File "adafruit_register/i2c_bit.py", line 70, in __set__
  File "adafruit_register/i2c_bit.py", line 70, in __set__
  File "adafruit_bus_device/i2c_device.py", line 104, in write
TypeError: extra keyword arguments given```

adafruit_bus_device/i2c_device.py#L104:
[self.i2c.writeto(self.device_address, buf, start=start, end=end, stop=stop)](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/2fd13d292dfa2f749f1ac46fd6b5e3cd61e8f4f1/adafruit_bus_device/i2c_device.py#L104)

tannewt, 2020-07-24:
Looks like we just need to remove it from the call and function signature. Libraries shouldn't use it anymore.

dhalbert, 2020-07-25:
Besides BusDevice, these libraries also still use or provide `stop=`:

```
ibraries/helpers/bitbangio/adafruit_bitbangio.py:124:    def writeto(self, address, buffer, *, start=0, end=None, st
op=True):
libraries/helpers/bitbangio/adafruit_bitbangio.py:151:        stop=True
libraries/helpers/bitbangio/adafruit_bitbangio.py:161:            self.writeto(address, buffer_out, start=out_start, 
end=out_end, stop=stop)

libraries/drivers/tca9584a/adafruit_tca9548a.py:100:                stop=False,
libraries/drivers/mlx90393/adafruit_mlx90393.py:165:                i2c.write(payload, stop=False)
```

anecdata, 2020-07-25:
#3206 and #3207 just need the underlying change, those libraries per se are OK. I suspect most I2C device libraries will fail due to bus_device, and most uses are `stop=True` or equivalent.

How long do we wait on "good first issue"? I'm happy to PR this bus_device library if there are no takers.

jerryneedell, 2020-07-25:
I can’t deal with this tonight. I can look at it tomorrow if it has not been resolved.

jepler, 2020-07-26:
I think I've filed issues for all the repos that were using stop=, according to Dan's list above

anecdata, 2020-07-26:
Fixed with https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/pull/55 merge

anecdata, 2020-07-26:
Fixed with #55 merge

closed 2020-07-26 by anecdata