adafruit/Adafruit_CircuitPython_BusDevice #56: Comma causes syntax error in i2c_device.py

phil-shenk, 2020-07-27:
Comma after last argument in 'write_then_readinto' function on line 113 throws syntax error on line 114 in i2c_device.py
python 3.5
proposed fix: delete comma (worked for us)

dhalbert, 2020-07-27:
The simplest test case is:
```python
def f(x,): pass      # works
def g(*, x,): pass    # raises SyntaxError
```
This was fixed in Python 3.6: https://bugs.python.org/issue9232: _Allow trailing comma in any function argument list._

We use the `black` code formatter, which by default inserts commas on all argument lines when the arguments are split across multiple lines, and we require that all code is formatted using `black`. So we'd have to reformat code in all our libraries using Python 3.5 syntax (`black --target-version py35`), to suppress this behavior. That would cause other issues with Python syntax we have that is past 3.5.

Are you able to upgrade past Python 3.5? I'm sorry this has happened to you, but I don't think we can accommodate this in all our libraries.

dhalbert, 2020-07-27:
OK, I'm sorry, I take this back. We _are_ using `black --target-version py35`: I had forgotten that.

dhalbert, 2020-07-27:
Fixed by #57.

closed 2020-07-27 by dhalbert