adafruit/Adafruit_CircuitPython_BusDevice #58: Have tsl2561 device at address 0x29 but the example code does not find the device?

Nodi-Rubrum, 2020-08-10:
Have tsl2561 device at address 0x29 but the example code does not find the device?

Tried...
`import board
import busio
import adafruit_tsl2561

i2c = busio.I2C(board.SCL, board.SDA)
tsl = adafruit_tsl2561.TSL2561(i2c)
tsl.lux
`

Error reported...
ValueError: No I2C device at address: 39

Output from i2cdetect...
`# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: UU -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- UU
`

caternuson, 2020-08-10:
The default address for the TSL2561 is `0x39`:
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561/blob/0f747d7822b2a008f6295a722f19a9051a906682/adafruit_tsl2561.py#L57

Since you are using a non-default address setting, you need to specify it:
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561/blob/0f747d7822b2a008f6295a722f19a9051a906682/adafruit_tsl2561.py#L82
ex:
```python
 tsl = adafruit_tsl2561.TSL2561(i2c, address=0x29)
```

closed 2020-08-10 by caternuson

Nodi-Rubrum, 2020-08-10:
Yup, actually I pulled a "TSL2591" out of inventory.   TSL2591 of course the default address is 0x29, where the TSL2561 is 0x39.