adafruit/Adafruit_CircuitPython_BusDevice #59: PCA9685 found in i2cdetect but throws [Errno 121] Remote I/O error

razvanphp, 2020-09-16:
We're trying a new devkit board for jetson nano (production edition): https://www.leopardimaging.com/product/nvidia-jetson-cameras/nvidia_nano_mipi_camera_kits/li-nano-cb/li-nano-cb/

It has an I2C header, we can see the address 40 in i2cdetect, but it throws errors in python:
[image]

```
WARNING: Carrier board is not from a Jetson Developer Kit.
WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
WARNING: and in fact is unlikely to work correctly.
Traceback (most recent call last):
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_bus_device/i2c_device.py", line 172, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/home/jetson/.local/lib/python3.8/site-packages/busio.py", line 116, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 49, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/home/jetson/.local/lib/python3.8/site-packages/Adafruit_PureIO/smbus.py", line 308, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_bus_device/i2c_device.py", line 178, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/home/jetson/.local/lib/python3.8/site-packages/busio.py", line 106, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 56, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/home/jetson/.local/lib/python3.8/site-packages/Adafruit_PureIO/smbus.py", line 179, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gst.py", line 30, in <module>
    pca = PCA9685(i2c, address = 0x40)
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_pca9685.py", line 137, in _init_
    self.i2c_device = i2c_device.I2CDevice(i2c_bus, address)
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_bus_device/i2c_device.py", line 68, in _init_
    self.__probe_for_device()
  File "/home/jetson/.local/lib/python3.8/site-packages/adafruit_bus_device/i2c_device.py", line 180, in __probe_for_device
    raise ValueError("No I2C device at address: %x" % self.device_address)
ValueError: No I2C device at address: 40
Exiting...
Cleaning up pins
```

We understand it's a custom board, but can you please at least point us in the right direction?

Thank you!
R

ladyada, 2020-09-16:
could be clock speed, flaky wiring. IOError is likely a hardware issue. use a scope to look at the traces.

razvanphp, 2020-09-16:
the code works on another nvidia dev board, so my guess is that SCL and SDA pins are not correctly detected, can this be? does adafruit code use Jetson.GPIO?

ladyada, 2020-09-16:
for i2c we talk directly to the kernel ioctls!