adafruit/Adafruit_CircuitPython_BusDevice #60: Problems running Neopixel code in CircuitPython 6.0 Beta.1 on Crickit Feather with STM32F405

sealyons, 2020-10-04:
Not exactly sure where to post this issue, so if it belongs somewhere else, please move.  The issue is I'm having problems running Neopixel code in CP 6.0 Beta.1 on Crickit Feather with STM32F405.  I'm running the 1003 MPY libraries.  First I get errors if I attempt to run the code given in the "Make It Glow with Crickit" learn guide.  I'm using a 60 LED/meter Neopixel strip and my Crickit is powered with a 5v/5A supply.  The code begins to run and I get some LEDs lighted but not correctly and after a few seconds I get the following error:

Traceback (most recent call last):
  File "code.py", line 63, in <module>
  File "code.py", line 26, in color_chase
  File "adafruit_seesaw/neopixel.py", line 157, in setitem
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 5] Input/output error

I then looked at the Cricket example code given and noticed that it was very different than what was shown in the "Make It Glow with Crickit" learn guide.  It's using _**crickit.init_neopixel()**_ to initialize and _**crickit.neopixel.fill(0)**_ to clear the strip and _**crickit.neopixel[#] = (r, g, b)**_ to set color of individual LEDs.  This is vastly different than the method calls used in the learn guide.  Not exactly sure which is the correct way this should be coded.

So I tried to write some code using the example code method to see if that would work and it appeared to work on first testing.  But upon further testing, it also caused tracebacks.  I get the following traceback running this code:

```
import time
from adafruit_crickit import crickit

def set_led(np, pos, color, secs):
    # Clear the strip.
    np.fill(0)
    # Set LED at defined position to color.
    np[pos] = color
    # Keep it on for secs.
    time.sleep(secs)
    return

# Strip of 60 NeoPixels.
crickit.init_neopixel(60)

np = crickit.neopixel

# Set mid-bright R, G, B colors.
colors = [(100, 0 , 0), (0, 100, 0), (0, 0, 100)]

# Move an alternating R,G,B LED down the strip (100 ms delay) and restart at the beginning when done.
pos = 0
color = 0
while True:
    set_led(np, pos, colors[color], 0.1)
    pos+=1
    if pos == 60: pos = 0
    color+=1
    if color == 3: color = 0
```
Traceback (most recent call last):
  File "code.py", line 21, in <module>
  File "code.py", line 5, in set_led
  File "adafruit_seesaw/neopixel.py", line 170, in fill
  File "adafruit_seesaw/neopixel.py", line 157, in setitem
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_seesaw/seesaw.py", line 451, in write
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 5] Input/output error

Again, all of this appears to be associated with the STM32F405 Feather running with the Crickit Feather.  Unfortunately I don't have another Feather that I can try to see if any of this goes away with another processor.  I am also in the process of trying to use the STM32 to directly drive the Neopixel strip (without the Crickit) to see if there are any issues when doing that.  That should show whether this something generic to the STM32 or the Crickit.  Since the errors seem to be pointing to seesaw, I'm guessing this is something related to the Crickit.  Please let me know if you need any additional info.

* Just a note, I ran this in Alpha.3 and got similar errors.

FoamyGuy, 2020-10-04:
I am able to replicate this issue using the code from the learn guide mentioned: https://learn.adafruit.com/make-it-glow-with-crickit/circuitpython#crickit-featherwing-2995042-5 on a Feather STM32405 and Feather Crickit.

tannewt, 2020-10-05:
This sounds like: https://github.com/adafruit/circuitpython/issues/3376

Let's close this and consolidate there.

closed 2020-10-05 by tannewt