adafruit/Adafruit_CircuitPython_BusDevice #77: Error 'WriteableBuffer' is not defined when using I2C device

bzuu-ic, 2022-02-11:
After updating to version 5.1.3 of this library I am getting the error: 

```
  File "<snip>/adafruit_bus_device/i2c_device.py", line 65, in I2CDevice
    self, buf: WriteableBuffer, *, start: int = 0, end: Optional[int] = None
NameError: name 'WriteableBuffer' is not defined
```

When looking at the `i2c_device.py` file I noticed the latest build for that file has failed as well with the same error, see:

https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/runs/5070616660?check_suite_focus=true

billvanleeuwen424, 2022-02-11:
Having this same issue upon updatng

billvanleeuwen424, 2022-02-11:
> I'm also having this issue after updating. Does anyone have any ideas or a fix? Kind of at a loss of what to do next and I need it for a critical application. I'll try downgrading now.
> 
> @bzuu-ic Thanks for reporting this. I am glad it isn't just me.

Downgrading worked for me. There is probably a more intuitive way to do it, but just copying and replacing an older version of i2c_device.py worked for me

bzuu-ic, 2022-02-11:
@billvanleeuwen424 Glad to hear you could solve it with a downgrade.

I am also trying to get it to work with a downgraded version, but am kind of stuck unfortunately with packaging with `shiv`. Although my `Pipfile.lock` explicitly states 5.1.2 it still ends up using 5.1.3 in the final package.

dhalbert, 2022-02-11:
Upgrade adafruit-blinka as well. That should fix this error:
```sh
pip3 install --upgrade adafruit-blinka
```

bzuu-ic, 2022-02-11:
@dhalbert Thanks, got it working again :)

closed 2022-02-11 by TheKitty

dhalbert, 2022-02-11:
Version 5.1.4 is now released and should fix this.