adafruit/Adafruit_CircuitPython_BusDevice #82: Dependency Issues With Jetson Nano

808brick, 2022-02-15:
Hi. 

I am trying to use the ADS1115 sensor in Python using the Python library:
[https://github.com/adafruit/Adafruit_Python_ADS1x15/](https://github.com/adafruit/Adafruit_Python_ADS1x15/ )

One of the dependencies for that project is this Adafruit_CircuitPython_BusDevice module. Whenever running through the tutorials on the Adafruit website for the module: 
[https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/python-circuitpython](https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/python-circuitpython)

I get the following Traceback error in Python 3.6.9 on a Jetson Nano 

```
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> import adafruit_ads1x15.ads1115 as ADS
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_ads1x15/ads1115.py", line 16, in <module>
    from .ads1x15 import ADS1x15, Mode
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_ads1x15/ads1x15.py", line 19, in <module>
    from adafruit_bus_device.i2c_device import I2CDevice
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 25, in <module>
    class I2CDevice:
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 55, in I2CDevice
    def __init__(self, i2c: I2C, device_address: int, probe: bool = True) -> None:
NameError: name 'I2C' is not defined
```
However, the I2C class, which comes from the busio modules, is able to import in just fine on my device. After some digging, it turns out a different import was actually failing, but not being reported because of a try-except wrapper around a series of import statements 
 in the i2c_device.py file. After some searching it would appear the actual module at fault is the circuitpython_typing import statement here:
[https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/main/adafruit_bus_device/i2c_device.py#L13](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/main/adafruit_bus_device/i2c_device.py#L13)

But because it came before the busio import statement, busio never got imported. 

I'm not sure why import statements would be placed between try-except statements without any type of error printing, that should be changed if possible, as it makes debugging dependency issues like this very difficult, and I don't think there would be a case which you would be okay with failing to import a dependency. 

I tried to install the circuitpython-typing module, as recommended on the [official GitHub repo ](https://github.com/adafruit/Adafruit_CircuitPython_Typing)
`pip3 install adafruit-circuitpython-typing`

But I am left with the following error:
```
ERROR: Could not find a version that satisfies the requirement adafruit-circuitpython-typing (from versions: none)
ERROR: No matching distribution found for adafruit-circuitpython-typing
```

I am able to run the sample code for the [CircuitPython Jetson Nano setup](https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup) just fine, so I am not sure why this particular module is posing an issue. 

Out of curiosity, I modified the order of the import statements in i2c_device.py, so busio got import prior to circuitpython_typing, as well as print out any ImportError exceptions, and got the following result (when trying to 

```
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> import adafruit_ads1x15.ads1115 as ADS
No module named 'circuitpython_typing'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_ads1x15/ads1115.py", line 16, in <module>
    from .ads1x15 import ADS1x15, Mode
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_ads1x15/ads1x15.py", line 19, in <module>
    from adafruit_bus_device.i2c_device import I2CDevice
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 27, in <module>
    class I2CDevice:
  File "/home/camera/.local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 66, in I2CDevice
    self, buf: WriteableBuffer, *, start: int = 0, end: Optional[int] = None
NameError: name 'WriteableBuffer' is not defined

``` 

So it seems the root of my problem is needing the circuitpython_typing dependency, but for some reason it is not available for the Jetson Nano? If anyone had any insight into this issue it would be greatly appreciated. 

If you instead think I should post this issue, on the CircuitPython_Typing GitHub page, then please let me know. I only placed it here since the errors are due to dependencies for the i2c_device.py file in this GitHub repo, and others may run into the same issue. 

-Raymond

sergio100899, 2022-02-15:
I have the same problem with the i2c_device.py code.

dhalbert, 2022-02-15:
This problem is fixed in the latest adafruit-blinka release, but because your Python is 3.6.9, and newer adafruit-blinka releases require Python 3.7 or later, pip does not update to the latest adafruit-blinka. Can you run a newer version of Python, either by upgrading the whole Linux, or installing a newer version of Python? Do not force it to be the system`python3` or `python` -- that might break things. But if you install it as another version of Python, invoked by, say, python3.8 (or used in a venv), and then reinstall the libraries and run your code using that version of Python.

Discussions on this:
https://forums.developer.nvidia.com/t/how-to-support-python3-8/168813/17
https://www.google.com/search?q=jetson+python+version+site:forums.developer.nvidia.com

sergio100899, 2022-02-15:
> This problem is fixed in the latest adafruit-blinka release, but because your Python is 3.6.9. Newer adafruit-blinka releases require Python 3.7 or later, so pip does not update to the latest adafruit-blinka. Can you run a newer version of Python, either by upgrading the whole Linux, or installing a newer version of Python? Do not force it to be the system`python3` or `python` -- that might break things. But if you install it as another version of Python, invoked by, say, python3.8 (or used in a venv), and then reinstall the libraries and run your code using that version of Python.
> 
> Discussions on this: https://forums.developer.nvidia.com/t/how-to-support-python3-8/168813/17 https://www.google.com/search?q=jetson+python+version+site:forums.developer.nvidia.com

Thanks for the solutions it works for me.

808brick, 2022-02-15:
I can confirm upgrading to Python version 3.7 helped solve the issue.  

On Ubuntu I ran the following commands to properly update Python and pip on the Jetson Nano:

```
sudo apt update
sudo apt install python3.7
python3 -m pip install --upgrade pip
```

Just double check that your `python3 --version` shows the new 3.7 version. If not you can check out [this stackoverflow post](https://stackoverflow.com/questions/43062608/how-to-update-alternatives-to-python-3-without-breaking-apt) on updating which python version gets run by default, and if you want to update all your pip packages from 3.6 to 3.7, you can check out [this stckoverflow post](https://stackoverflow.com/questions/2720014/how-to-upgrade-all-python-packages-with-pip)

Thanks @dhalbert for your help.

closed 2022-02-15 by 808brick