adafruit/Adafruit_CircuitPython_BusDevice #83: name 'I2C' is not defined

sergio100899, 2022-02-15:
when i execute this line of code --from adafruit_servokit import ServoKit--, it gives me this error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/adafruit_servokit.py", line 36, in <module>
    from adafruit_pca9685 import PCA9685
  File "/usr/local/lib/python3.6/dist-packages/adafruit_pca9685.py", line 41, in <module>
    from adafruit_bus_device import i2c_device
  File "/usr/local/lib/python3.6/dist-packages/adafruit_bus_device/i2c_device.py", line 25, in <module>
    class I2CDevice:
  File "/usr/local/lib/python3.6/dist-packages/adafruit_bus_device/i2c_device.py", line 55, in I2CDevice
    def __init__(self, i2c: I2C, device_address: int, probe: bool = True) -> None:
NameError: name 'I2C' is not defined

I don´t know if i2c_device.py of the Adafruit_CircuitPython_BusDevice library has an error.

BjornS98, 2022-02-15:
Same here
[image]

FoamyGuy, 2022-02-15:
Can you post the results of running this command here:
```
pip show Adafruit_Blinka
```
This will show us the version of Blinka in use. I believe the issues you're seeing stem from an out of date version.

The currently released version of Blinka is `6.20.4` please ensure you're using that version and let us know if you still have any issues.

sergio100899, 2022-02-15:
> 

Thanks for the solution, the problem was the version of Blinka i had the 6.15 version. The 6.20.4 version works on newer versions of python and the default version of the jetson nano is 3.6 so first i had to update my python (in my case a installed 3.8) and then a i could update Blinka and my code now runs perfectly.

closed 2022-02-15 by sergio100899

BjornS98, 2022-02-23:
Easier solution is to downgrade the busdevice, and if needed blinka. That way you don't have to upgrade and you can import I2C as it should. Cause not all device (Jetson nano) like an upgrade in python --V. So just getting it out there.