adafruit/Adafruit_CircuitPython_BusDevice #87: NameError after update of libraries

a-schneider-fmi, 2022-02-23:
After updating the Adafruit CircuitPython libraries, a program that previously worked now throws a `NameError` exception when importing the library:
```
Traceback (most recent call last):
  File "measure_battery.py", line 15, in <module>
    import adafruit_ina260
  File "/home/pi/src/adafruit/Adafruit_CircuitPython_INA260/adafruit_ina260.py", line 27, in <module>
    import adafruit_bus_device.i2c_device as i2cdevice
  File "/home/pi/src/adafruit/Adafruit_CircuitPython_BusDevice/adafruit_bus_device/i2c_device.py", line 25, in <module>
    class I2CDevice:
  File "/home/pi/src/adafruit/Adafruit_CircuitPython_BusDevice/adafruit_bus_device/i2c_device.py", line 55, in I2CDevice
    def __init__(self, i2c: I2C, device_address: int, probe: bool = True) -> None:
NameError: name 'I2C' is not defined
```
I'm not sure if the problem resides in the BusDevice library that throws the error or the INA260 library.

FoamyGuy, 2022-02-23:
I believe this is something that has been fixed in newer versions of BusDevice and Blinka libraries. Can you post the versions of those two modules that you are using? You can find them with these commands:
```
pip show adafruit-blinka
pip show adafruit-circuitpython-busdevice
```

Also which version of Python are you using on your system? You can find that:
```
python --version
```

a-schneider-fmi, 2022-02-23:
I have cloned the libraries from git. Adafruit Blinka is on commit bd30f6a64bb61b7778fde4690428e5628d0bb725 and Adafruit CircuitPython Busdevice is on commit b96f0d279481446017ca1cc1ff4abf1f5a10346f. The system has Python 3.9.2 installed.

dhalbert, 2022-02-23:
You might have some old versions on your `PYTHONPATH`, perhaps from a previous pip install, that are being imported instead of the clones you are using. I just tried this on an RPi 4, without seeing the error you are seeing.
```sh
$ pip3 install --upgrade adafruit-blinka adafruit-circuitpython-busdevice adafruit-circuitpython-ina260
...
$ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_ina260
>>> import adafruit_bus_device.i2c_device as i2cdevice
>>> 

$ pip3 list |grep -i Adafruit
Adafruit-Blinka                  7.0.1
adafruit-circuitpython-busdevice 5.1.8
adafruit-circuitpython-ina260    1.3.4
adafruit-circuitpython-register  1.9.8
adafruit-circuitpython-typing    1.0.0
Adafruit-PlatformDetect          3.18.0
Adafruit-PureIO                  1.1.9
```

FoamyGuy, 2022-02-23:
I think you need to install [Adafruit_CircuitPython_Typing](https://github.com/adafruit/Adafruit_CircuitPython_Typing) as well. If that were missing it would raise an import error causing `I2C` not to be imported which is what your error is reporting.

`pip` might make it easier to manage the dependencies because it will install them automatically when needed. If you want to do it manually you have to make sure that you check in `requirements.txt` of the libraries and clone / install anything you find in there.

a-schneider-fmi, 2022-02-23:
After cloning [Adafruit_CircuitPython_Typing](https://github.com/adafruit/Adafruit_CircuitPython_Typing) as well, it indeed works. Thanks @FoamyGuy for the hint. I have overlooked that the requirements have changed. Sorry for the noise.

closed 2022-02-23 by a-schneider-fmi