adafruit/Adafruit_CircuitPython_BusDevice #88: failed to "pip3 install adafruit-circuitpython-busdevice"

ztbupt, 2022-05-06:
the log is below:
Collecting adafruit-circuitpython-busdevice
Collecting adafruit-circuitpython-typing (from adafruit-circuitpython-busdevice)
Collecting Adafruit-Blinka>=6.20.4 (from adafruit-circuitpython-busdevice)
  Could not find a version that satisfies the requirement Adafruit-Blinka>=6.20.4 (from adafruit-circuitpython-busdevice) (from versions: 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.1.7, 0.1.8, 0.1.9, 0.1.10, 0.2, 0.2.1, 0.2.2, 0.2.3, 0.2.4, 0.2.5, 0.2.6, 0.2.7, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 1.0.2, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.2.3, 1.2.4, 1.2.5, 1.2.6, 1.2.7, 1.2.8, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 2.0.0, 2.0.1, 2.0.2, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.1.5, 2.1.6, 2.2.0, 2.2.1, 2.3.0, 2.3.1, 2.3.2, 2.3.3, 2.4.0, 2.4.1, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.6.0, 2.6.1, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.0.4, 3.0.5, 3.0.6, 3.0.7, 3.0.8, 3.1.0, 3.1.1, 3.1.2, 3.2.0, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.3.5, 3.3.6, 3.3.7, 3.3.8, 3.3.9, 3.3.10, 3.4.0, 3.4.1, 3.5.0, 3.5.1, 3.5.2, 3.6.0, 3.6.1, 3.7.0, 3.7.1, 3.8.0, 3.9.0, 3.10.0, 4.0.0, 4.1.0, 4.2.0, 4.3.0, 4.4.0, 4.5.0, 4.6.0, 4.7.0, 4.8.0, 4.8.1, 4.9.0, 4.10.0, 4.10.1, 5.0.0, 5.0.1, 5.1.0, 5.2.0, 5.2.1, 5.2.2, 5.2.3, 5.2.4, 5.3.0, 5.3.1, 5.3.2, 5.3.3, 5.3.4, 5.4.0, 5.4.1, 5.5.0, 5.5.1, 5.5.2, 5.5.3, 5.5.4, 5.6.0, 5.7.0, 5.8.0, 5.8.1, 5.8.2, 5.9.0, 5.9.1, 5.9.2, 5.10.0, 5.11.0, 5.12.0, 5.13.0, 5.13.1, 6.0.0, 6.0.1, 6.0.2, 6.1.0, 6.2.0, 6.2.1, 6.2.2, 6.3.0, 6.3.1, 6.3.2, 6.4.0, 6.4.1, 6.4.2, 6.5.0, 6.6.0, 6.6.1, 6.6.2, 6.7.0, 6.8.0, 6.8.1, 6.8.2, 6.8.3, 6.9.0, 6.9.1, 6.9.2, 6.10.0, 6.10.1, 6.10.2, 6.10.3, 6.11.0, 6.11.1, 6.12.0, 6.13.0, 6.13.1, 6.14.0, 6.14.1, 6.15.0)
No matching distribution found for Adafruit-Blinka>=6.20.4 (from adafruit-circuitpython-busdevice)

Could you help to fix it?

closed 2022-05-20 by tekktrik

camilofernandez9405, 2022-11-19:
[image]

Hola comunidad quisisera si me pueden ayudar con este error no se que hacer se los agradeceria de corazon estoy utilizando jetpack 4.6

tekktrik, 2022-11-19:
You will need to upgrade to Python 3.7 to use the latest version of the library since it uses Blinka 7.0.  You could use an older version of the library if you don't want to upgrade.

camilofernandez9405, 2022-11-19:
podría orientarme de como hacerlo, se lo agradecería mucho un abrazo

tekktrik, 2022-11-19:
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup

camilofernandez9405, 2022-11-19:
vale ya pude solucionar el error, ahora el problema que tengo es que no puedo instalar opencv para esa version de python de 3.7 y con la anterior version 3.6.9 si lo habia podido instalar con el comando sudo apt-get install python3-opencv, pero ahora no con la nueva version