adafruit/Adafruit_CircuitPython_BusDevice #100: Add support for I2C General Call

caternuson, 2024-06-25:
The current set of I2C read/write methods in `I2CDevice` all use the device address that is specified when the `I2CDevice` instance is created. This is by design and generally makes I2C comms easier. However, I2C General Calls are done by writing to the reserved address `0x00`. One example of using a General Call is for a Software reset (taken from the [I2C spec](https://www.nxp.com/docs/en/user-guide/UM10204.pdf)):
[image]

One example of a device that supports this is the MCP4728:
https://www.adafruit.com/product/4470
and the sending of the device address prevents a true General Call from occurring, so the software reset does not occur:
https://github.com/adafruit/Adafruit_CircuitPython_MCP4728/issues/18

Since the General Call is an I2C feature, and not a device specific feature, it seems like supporting General Call should be added to `I2CDevice`. Maybe a new `general_call()` method that simply does a write to the reserved `0x00` address?
```python
with self.i2c_device as i2c:
    i2c.general_call(some_byte_array)
```

tannewt, 2024-06-26:
I think it'd be better to just use the I2C bus directly instead of I2CDevice. The general call will impact all devices, not just the one represented by the I2CDevice object.

caternuson, 2024-06-26:
Something like this?
```python
with self.i2c_device as i2c:
    i2c.i2c.writeto(0x00, some_byte_array)
```
So the context manager could still be used for bus locking/unlocking.

tannewt, 2024-06-27:
Sure, that's fine.

caternuson, 2024-06-28:
OK, going to close this. Drivers can take approach above if they need to make a General Call.

closed 2024-06-28 by caternuson